rust-embedded/cortex-m #340: v0.7.2 doesn't build on Apple Silicon

john-terrell, 2021-04-23:
Trying to compile 0.7.2 on an Apple Silicon Mac fails.   Current master branch compiles correctly (due to a9fb68728a0b98b98da3e4215670b2f1c1b6b982) so it appears a new release needs to be published to crates.io.

adamgreig, 2021-04-24:
Is this causing you an actual problem? Normally you don't need to build cortex-m for the host platform; it's pretty much only supported for when docs.rs couldn't be configured to use other platforms, to allow docs to be built on x86. You should still be able to build for Cortex-M targets fine, though, which is why we haven't made a release just for this issue.

john-terrell, 2021-04-24:
@adamgreig  - Oh jeeze...here's my "I'm dumb face":  :-(.   I was just building it natively and not considering what was happening.   Sorry for the spam.

closed 2021-04-24 by john-terrell

adamgreig, 2021-04-24:
No worries!