rust-embedded/cortex-m #343: Delay: Don't rely on EH traits

David-OConnor, 2021-06-12:
File in question: https://github.com/rust-embedded/cortex-m/blob/1a4e09646aa37c3408d69fe4452265f405400593/src/delay.rs#L32

Typically, HALs implement their own Delay. Given it only uses cortex-m peripheral, there's no reason HALs need to keep this; this `cortex-m` library includes a delay.

However, it's dependent on `embedded-hal::Delay`. Can we make the implementation native, and add EH as an optional add-on? This way, you don't need to import the trait, or include it in your project. I'll do a PR if approved.

David-OConnor, 2021-06-12:
In case it helps, the use case is this:

- I have a STM32 HAL library that implements its own Delay (In a similar way to other cortex-m-based HAL libraries). It uses the cortex_m systick peripheral, and no stm32-specific peripherals. I think this, and other HALs can delete their Delay modules, and direct users to use the one here, in `cortex_m` instead.

- I don't think user code should rely on `embedded-hal` traits; their main use case is in writing generic drivers. Currently, I'm unable to delete the `Delay` module and defer to the `cortex-m` crate, since it would require user code to import `embedded-hal`, since `cortex_m::delay::Delay` requires it.

Disasm, 2021-06-12:
I still don't get why `cortex-m` Delay driver shouldn't rely on `embedded-hal` while HALs can rely on `embedded-hal` for the same delay driver.

David-OConnor, 2021-06-12:
HALs shouldn't IMO. Generic drivers should.

Disasm, 2021-06-12:
Yes, and which crate should provide these `DelayMs/DelayUs` implementations for generic drivers?

David-OConnor, 2021-06-12:
The HALs provide them. The drivers use them, and import the trait.

User code that flashes to an MCU doesn't need them.

@Rahix  has a good [write up here](https://github.com/rust-embedded/embedded-hal/issues/177#issuecomment-821083026)

Disasm, 2021-06-12:
So you suggest that `cortex-m` has a Delay driver, but DelayMs/DelayUs should be provided by HALs? I don't understand why it should work like this. What if I have a generic Cortex-M (FPGA) core and no HAL?

David-OConnor, 2021-06-12:
No - I'm suggesting `cortex-m` supply a native impl, and an EH impl - as in the PR. And HALs delete their Delay module.

Disasm, 2021-06-12:
Ohh, I forgot that now we have fallible-only `Delay*` traits. Now it makes sense.

commit 47246ecc (2021-06-25): Merge #344

344: Remove dependence on embedded-hal for Delay r=adamgreig a=David-OConnor

Fixes #343

No functional changes. Allows you to use Delay as previously, or without including or importing `embedded-hal`.
PR #344 (2021-06-25): Remove dependence on embedded-hal for Delay
Fixes #343

No functional changes. Allows you to use Delay as previously, or without including or importing `embedded-hal`.
closed 2021-06-25 by bors[bot]