rust-embedded/cortex-m #364: singleton!()'s use of Option can have nasty optimization resullts

Rahix, 2021-11-18:
I was trying to store some large buffers inside of a `cortex_m::singleton!()` when I noticed that suddenly my flash usage increased dramatically.  As some (contrived) examples, the following works (variable ends up in `.bss`):

```rust
cortex_m::singleton!(: [u8; 2048] = [0xff; 2048]);
```

but this doesn't (variable ends up in `.data`):

```rust
cortex_m::singleton!(: [bool; 2048] = [true; 2048]);
```

The reason is the `Option` inside the `singleton!()` macro:

https://github.com/rust-embedded/cortex-m/blob/6b013138b734b9bbeb24a345f75d2bcc1c69fa8d/src/macros.rs#L55

Normally, the `None` value which is stored here has a bitpattern of all zeros, leading to it being stored in `.bss`.  But when the given type has a niche (like `bool` above does), the `None` variant is "optimized" into it.  This means it then has non-zero bits and cannot be stored in `.bss` anymore, leading to it getting moved to `.data`...

A quick solution is to "reimplement" the `Option`:

```rust
	static mut VAR: (MaybeUninit<$ty>, bool) = (MaybeUninit::uninit(), false);
```

This will always end up in `.bss` because no niche optimization exists for such types (yet).  Before opening a PR, I wanted to discuss whether there are any objections to changing to such an implementation?

Side Note: I think it would be nice to allow users to overwrite the name of the `static`.  When looking at `nm` or `bloaty` output, only seeing a bunch of types named `__cortex_m_rt_main::{{closure}}::VAR` is not great...

TDHolmes, 2021-12-30:
This change seems reasonable to me if you can show that it fixes the performance issue you found and has similar performance as the existing implementation. `MaybeUninit` makes a lot of sense for this macro, so seems reasonable to me...

Also a variant of this macro where we pass in a name seems nice

TDHolmes, 2021-12-30:
here's what I came up with. It unfortunately involves adding a dependency `paste` to get the optional name, but it's a lightweight dependency (no transitive dependencies). It does require adding an atomic bool to indicate if we're initialized or not though, but that isn't too much overhead imo. `Option` would often have a similar overhead.

```rust
macro_rules! singleton {
    ($name:ident : $ty:ty = $expr:expr) => {
        $crate::interrupt::free(|_| {
            use ::core::{
                mem::MaybeUninit,
                sync::atomic::{AtomicBool, Ordering},
            };

            paste::paste! {
                static mut [< $name >]: MaybeUninit<$ty> = MaybeUninit::uninit();
                static [<$name _INITIALIZED>]: AtomicBool =
                    AtomicBool::new(false);

                if [<$name _INITIALIZED>].load(Ordering::Acquire) {
                    None
                } else {
                    let expr = $expr;

                    // initializes our T and returns an &mut to it
                    let t = unsafe { [< $name >].write(expr) };
                    [<$name _INITIALIZED>].store(true, Ordering::Acquire);

                    #[allow(unsafe_code)]
                    unsafe { Some(t) }
                }
            }
        })
    };
    (: $ty:ty = $expr:expr) => {
        // The old way - use the name VAR
        singleton!(VAR: $ty = $expr)
    };
}
```

Rahix, 2022-01-01:
> fixes the performance issue you found

This isn't about performance, but about flash usage.  If the static ends up in `.data`, it potentially wastes a huge amount of flash.  It should really be in `.bss` where it only occupies space in RAM at runtime.

> has similar performance as the existing implementation

I did not check but I would be surprised if a boolean flag vs an `Option` ends up generating much different code.

> It unfortunately involves adding a dependency paste to get the optional name, but it's a lightweight dependency (no transitive dependencies).

You can avoid this by just putting both variables into one static as I showed above:

```rust
static mut VAR: (MaybeUninit<$ty>, bool) = (MaybeUninit::uninit(), false);
```

> It does require adding an atomic bool to indicate if we're initialized or not though

This is unnecessary.  Interrupts are disabled already and the previous code also does not do any kind of synchronization on the `Option` so no atomic is needed in a new version either.

---

For reference, in the project where this came up, I am using the following macro as a replacement:

```rust
#[macro_export]
macro_rules! singleton {
    ($name:ident: $ty:ty = $expr:expr) => {
        ::cortex_m::interrupt::free(|_| {
            static mut $name: (::core::mem::MaybeUninit<$ty>, bool) =
                (::core::mem::MaybeUninit::uninit(), false);

            #[allow(unsafe_code)]
            let used = unsafe { $name.1 };
            if used {
                None
            } else {
                let expr = $expr;

                #[allow(unsafe_code)]
                unsafe {
                    $name.1 = true;
                    Some($name.0.write(expr))
                }
            }
        })
    };
}
```

Rahix, 2022-01-01:
Unrelated side-note about your code:  If you are going for atomics instead of a critical section, the code is racy:  You must access the atomic in a single operation like [`AtomicBool::compare_exchange()`](https://doc.rust-lang.org/core/sync/atomic/struct.AtomicBool.html#method.compare_exchange) instead of your

```rust
if some_atomic.load(Ordering::Acquire) {
    ...
} else {
    ...
    some_atomic.store(true, Ordering::Release);
    ...
}
```

Otherwise a racing task could run between the load and store.

---

But again, this is not relevant to this issue because we're in a critical section and the atomic is not necessary here anyway.

TDHolmes, 2022-01-01:
By performance, I mainly meant WRT flash usage, but I did kind of mix and match what I meant by the word.

Ah, didn't think to use a tuple to get around having a separate variable tracking initialization. Nice!

Yeah I wasn't thinking of switching out the critical section for an atomic, just didn't fully think that one through. I don't think CAS is available on thumbv6m so not possible anyways.

Now if you add the existing match arm with VAR as the name I think this could be a backwards compatible change

Rahix, 2022-01-03:
Sounds good, will send a PR.

closed 2022-01-04 by bors[bot]