rust-embedded/cortex-m #376: DWT: differentiate the first comparator

tmplt, 2022-01-01:
As discussed in #364 we should ideally check whether a DWT feature is supported statically. This is only realistically applicable for cycle count comparisons (but still requires a runtime check) which is only supported on the first comparator. An implementation would then have some `DWTComparatorFirst` and `DWTComparator` structures that implement a common trait: if cycle count comparison is configured for `DWTComparator`, an error is returned.

Opening this issue for discussion purposes for now. At present the documentation warns that cycle count comparison is only available with the first comparator.

tmplt, 2022-01-01:
CC @TDHolmes

TDHolmes, 2022-01-01:
Thanks for opening this so we don’t forget. IIRC, the second comparator is
also special (specific for data compare? I don’t recall the exact details)

TDHolmes, 2022-01-01:
🏷️ #367 (the cycle count PR)

TDHolmes, 2022-01-02:
I was wrong, only comparator zero is special. Opened up a PR