rust-embedded/cortex-m #406: Make the `HardFault` trampoline opt-in

jonas-schievink, 2022-01-23:
Currently, *any* user of this crate has to pay for a small Assembly trampoline that provides access to the `&ExceptionFrame` to the `HardFault` handler (even if they don't use the `&ExceptionFrame`, or don't even define any `HardFault` handler):

https://github.com/rust-embedded/cortex-m-rt/blob/492c7bf6212486b378ecf0518572d180a50c83a1/asm.s#L8-L19

Since few people make use of the `&ExceptionFrame` argument, we could add a Cargo feature to turn it on instead, and generate the right linker script in `build.rs`. By forwarding the feature to the macro crate, `#[exception]` can provide good diagnostics when trying to use `&ExceptionFrame` without the feature enabled.

I am planning on extending the trampoline mechanism to other faults and maybe even all exceptions, which I've prototyped by making the ASM trampoline generic and have it work for all faults. This makes it larger and adds more data to the excutable, so making this opt-in would help reduce the cost. The trampoline code could also benefit from ARMv7/v8-specific improvements, which I'll work on later, which should shrink it again (for the higher platforms at least).

I suppose we could also have one Cargo feature per exception. The build script would then generate a linker script that pulls in exactly the trampolines that are needed. This should even still work with the precompiled `.a` files, if we do it right (putting every trampoline in its own section and making sure they're only pulled in when necessary).

Initially, defining 10 or 11 Cargo features for this sounds pretty sketchy, but most apps won't use any of them, and even "advanced" apps that implement an RTOS can get away with just 1 or 2 (one for SVCall to read syscall arguments, and one on HardFault/DefaultHandler for debugging crashes).

An ideal solution would not require Cargo features at all and instead make the `#[exception]` macro pull in the Assembly trampoline. I have not found any way to do this though. If one does come up (perhaps using the currently unstable [`#[link_args]`](https://github.com/rust-lang/rust/issues/29596) attribute), we can make the Cargo features no-ops until the next major version.

cbiffle, 2023-01-25:
@adamgreig - I believe this was closed inadvertently after being merely mentioned on #423. (Since I still can't work out how to turn off the trampoline without replacing the linker script.)

adamgreig, 2023-01-25:
I'm still seeing it as open, but it certainly shouldn't be closed yet, you're right.

Now that we have cfg_global_asm being used with features like set-vtor and set-sp, and soon the zero-init-ram in #455, I wonder if we can use a similar trick to conditionally include/remove the trampoline, which at least wouldn't be a breaking change (unlike just removing it entirely).

cbiffle, 2023-01-25:
You're right, I appear to have mistaken a PR merge icon for an Issue Closed icon! My mistake.

And, yeah, I assume someone out there is using it, so removing it entirely seems aggressive.

SCingolani, 2023-05-03:
Hi, so was some agreement reached regarding what to do with this issue? I am facing a situation where the current implementation is not suitable (i.e. I need to mutate the ExceptionFrame so that the PC points to the instruction that comes after the one that triggered the exception, as well as having access to the SP at the moment the HardFault is entered so I can pop the EXC_RETURN). For my case, either the implementation would need to be slightly modified, or as suggested, this should be made optional / removed from the crate, so I can use a custom handler...

closed 2023-08-16 by adamgreig