rust-embedded/cortex-m #411: statics defined in #[entry] are not safe on multi-core systems

Disasm, 2022-01-23:
RP2040 starts execution on both cores. If entry function contains a "special" definition of `static mut` variable, each core takes a mutable reference to it, which is illegal from the Rust point of view.

Here is the relevant piece of `#[entry]` code:
https://github.com/rust-embedded/cortex-m-rt/blob/ca4790f40738bcc770285c8080955a2077682078/macros/src/lib.rs#L87-L88

thejpster, 2024-02-06:
It's unsound where the interrupt function can be re-entered. The NVIC usually prohibits this but on a multi-core system you need something at the HAL level to ensure this. You can imagine a mechanism that moves ownership of system interrupts from one core to another, hiding the regular NVIC API from the user. Then it would still be sound.

9names, 2024-09-14:
> The NVIC usually prohibits this but on a multi-core system you need something at the HAL level to ensure this.

You can also re-enable interrupts inside an interrupt and get the same problem.

In spite of rp2040 being around for several years we haven't had a single report of anyone having issues with this.

Rather than trying to make this safe, I think we should just document that you should not do anything that allows concurrent/recurrent execution of interrupt handlers and leave it at that, since the actual real-world usages of this behaviour are exceptionally rare, and even in the absence of the static mut transform are likely to be unsafe.

jannic, 2024-09-14:
> In spite of rp2040 being around for several years we haven't had a single report of anyone having issues with this.

I'm not sure if this is a strong argument. Having two instances of `&mut` references to the same underlying static is instant UB, but that doesn't guarantee that something bad happens. And even if it somehow breaks the software, it may happen very occasional, basically impossible to reproduce, and with non-obvious symptoms. So it may be dismissed as a flaky device and never be traced back to the underlying cause.

IMHO if we know that there are conditions that the user must hold up to avoid UB, there should be at least one `unsafe` function with a useful safety comment, so the user has a fair chance of knowing about these requirements, and a point to start looking for UB in case the software doesn't work as expected.

thejpster, 2024-09-14:
It should go behind a flag, like the cs impl. Maybe even the same flag.