rust-embedded/cortex-m #419: Hardcoded path name for

noppej, 2022-02-01:


TDHolmes, 2022-02-01:
This will soon be irrelevant as I believe this path for making assembly will be eliminated

noppej, 2022-02-01:
Excellent. Do you have a reference to a discussion that I can follow for this please?

TDHolmes, 2022-02-01:
I'm not aware of one yet. Once 1.59 releases with stable support for global asm the feature inline-asm will be made the default / only option. You could switch to using that feature in the meantime

adamgreig, 2022-02-01:
At the moment the inline-asm feature doesn't change how the cortex-m-rt asm is generated, but the plan is indeed to move it to use `global_asm!()` for Rust 1.59, at which point `asm.S` can be deleted entirely.

If there was some easy fix we could apply now, I'd be happy to merge it, but I don't want to spend too long investigating when we plan to change the approach entirely once the next Rust release is out.

noppej, 2022-02-01:
Totally understand, and I can easily wait for Rust 1.59. Thanks for the info.

closed 2022-02-01 by noppej

adamgreig, 2022-02-02:
See #420 for progress.