rust-embedded/cortex-m #443: 0.7.5 release appears to still depend on bare-metal 0.2.5

workingjubilee, 2022-06-21:
Unfortunately the recent 0.7.5 release was cut from the staging branch.
https://github.com/rust-embedded/cortex-m/blob/e0bfe3ae21903e9dbd80e903e726f7341662e12b/Cargo.toml#L14-L22

Thus, it did not include recent dependency updates that are present on other branches, so these risk needless duplication when depending on this crate and also other ecosystem crates that do use the latest. It is not a critical priority for me, but I thought you might want to know.

workingjubilee, 2022-06-23:
Hmm. Formally speaking, SemVer specifies,
> Major version zero (0.y.z) is for initial development. Anything MAY change at any time. The public API SHOULD NOT be considered stable.

and that
> How do I know when to release 1.0.0?
> 
> If your software is being used in production, it should probably already be 1.0.0. If you have a stable API on which users have come to depend, you should be 1.0.0. If you’re worrying a lot about backwards compatibility, you should probably already be 1.0.0.

But obviously it is not my crate, I just enjoy that sort of trivia.

In a more practical sense, I think "release candidate" versions are useful for propagating experimental releases that haven't quite finished baking yet but the ecosystem would find useful to build off of.

adamgreig, 2022-06-23:
Cargo uses a [slightly variant definition of semantic versioning](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html#specifying-dependencies-from-cratesio):

> This compatibility convention is different from SemVer in the way it treats versions before 1.0.0. While SemVer says there is no compatibility before 1.0.0, Cargo considers 0.x.y to be compatible with 0.x.z, where y ≥ z and x > 0.

In effect, everything in the 0.7 branch must remain compatible, so we save breaking changes like this for 0.8. But yes, we'd like to put out an alpha release of 0.8 soon which would include this.

closed 2024-04-16 by adamgreig