rust-embedded/cortex-m #445: Override/Discard the provided exception table

bhav97, 2022-07-18:
Hello
I'm working on a XMC1100 Cortex-M0 device which stores some clock initialization information for the startup software inside the unused values of the exception table.

Going through the cortex-m-rt source code, The reserved values in __EXCEPTIONS do not look like they can be overridden in any way. I tried duplicating the exception table with the required values and added a /DISCARD/ section to my memory.x, but I cannot seem to get it working.

Current memory.x
```
MEMORY
{
    FLASH   (RX)    : ORIGIN = 0x10001000, LENGTH = 64K
    RAM     (!RX)   : ORIGIN = 0x20000000, LENGTH = 16K
}

_stack_start = ORIGIN(RAM) + LENGTH(RAM);

SECTIONS
{
    /DISCARD/ :
    {
        *cortex_m_rt* (__EXCEPTIONS)
    }
}
```

Is it possible to override or discard just the exception table from the cortex-m-rt? Or should I just link against a modified copy of link.x?

adamgreig, 2022-07-19:
There's no problem with linking against a project-local copy of `link.x`, and that's probably the best way to go for now. If you can see any obvious way we could extend the default `link.x` to permit overriding these from `memory.x` without needing a custom `link.x`, I'd be interested to hear!

adamgreig, 2022-07-19:
You might want to check out https://github.com/david-sawatzke/xmc1100-hal which appears to use a custom linker script in https://github.com/david-sawatzke/xmc1100-hal/blob/master/xmc.x.in#L76 to accomplish this (and perhaps would be useful for you anyway).

bhav97, 2022-07-20:
Linking with a project local `link.x` sounds like the way to go for now. I thought I could be sneaky with the DISCARD and skip duplicating  the `link.x` .

closed 2022-07-20 by bhav97