rust-embedded/cortex-m #466: How to configure the NVIC registers

DmitriLyalikov, 2023-02-12:
Hi guys, 

I am relatively new to working with the cortex-m crate. 
I am working on debugging a spi slave implementation and would like to check whether or not my SPI0_IRQ is enabled in the ISER register, and if not, I will enable it.

I based my work off the examples/device.rs project,
where the following does what I would desire.
`let nvic = p.NVIC;
nvic.enable(Interrupts::SPI0_IRQ);`

However, i find that the function "enable" does not exist.
I am working in the RTIC environment with an RP2040. 

I try:
`        let mut nvic = c.core.NVIC;
        nvic::enable(Interrupt::SPI0_IRQ);`

Is there a new way to interact with the NVIC, or am I missing something? 
Thanks in advance

rjsberry, 2023-02-27:
In v0.7 of `cortex-m`, you can use [`NVIC::is_enabled`](https://docs.rs/cortex-m/latest/cortex_m/peripheral/struct.NVIC.html#method.is_enabled) and [`NVIC::unmask`](https://docs.rs/cortex-m/latest/cortex_m/peripheral/struct.NVIC.html#method.unmask).

These methods don't require an _instance_ of the NVIC peripheral (e.g., after calling [`Peripherals::take`](https://docs.rs/cortex-m/latest/cortex_m/peripheral/struct.Peripherals.html#method.take)).

```rust
use cortex_m::peripheral::NVIC;
use rp2040_pac::Interrupts;

if !NVIC::is_enabled(Interrupts::SPI0_IRQ) {
    unsafe { NVIC::unmask(Interrupts::SPI0_IRQ) };
}
```

Double check the safety docs for [`NVIC::unmask`](https://docs.rs/cortex-m/latest/cortex_m/peripheral/struct.NVIC.html#method.unmask) to make sure your use of `unsafe` is sound.

jonah-saltzman, 2023-03-13:
Is `NVIC::unmask` the equivalent of `__NVIC_EnableIRQ` in the ARM CMSIS abstraction layer?

newAM, 2024-06-30:
I'm triaging Cortex-M issues today.

I think the original question has been resolved, feel free to re-open if not!

As for `NVIC::unmask` vs `__NVIC_EnableIRQ`, I don't think they are entirely the same:

* https://docs.rs/cortex-m/0.7.7/src/cortex_m/peripheral/nvic.rs.html#115-122
* https://github.com/ARM-software/CMSIS_4/blob/f2cad4345783c948ed4a7f5cdb02cdc0856366f1/CMSIS/Include/core_cm0.h#L631

closed 2024-06-30 by newAM