rust-embedded/cortex-m #475: rust-lld: warning: section type mismatch for .got

Dirbaio, 2023-04-12:
Every project using `cortex-m-rt` gets these linker warnings. I don't know what could cause them, it might be good to investigate if we can get rid of them.

Normally they're not visible (cargo or rustc doesn't show linker output if it's successful). They're only visible when you get an actual real linker error. The linker error gets often printed *after* these warnings. I've seen them confuse people a few times on Matrix, thinking the warnings are what's causing the linker failure and trying to troubleshoot them.

```
          rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
 
          rust-lld: warning: section type mismatch for .got.plt
          >>> <internal>:(.got.plt): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
 
          rust-lld: warning: section type mismatch for .got
          >>> <internal>:(.got): SHT_PROGBITS
          >>> output section .got: SHT_NOBITS
```

JosefUtbult, 2023-04-18:
I have the same problem when using cortex-m and usb-device.

JosefUtbult, 2023-04-18:
I got some help which seems to have solved my problem. Try adding the following feature in Cargo.toml

```critical-section-single-core```

Dirbaio, 2024-04-16:
Rust 1.73 (probably due to the upgrade to llvm 17) has magically fixed this warning for us. It doesn't show up with latest Rust versions anymore.

closed 2024-04-16 by Dirbaio