rust-embedded/cortex-m #481: Cannot build crate with ['Inline-asm"] feature

Takkaryx, 2023-07-04:
Hello all, I'm trying to get into embedded development with rust.  I'm trying to use the inline-asm feature, but it will not compile.

the cargo.toml line is:
`cortex-m = { version = "0.7.6", features = ["inline-asm", "critical-section-single-core"] }` (note, I tried 0.7.7 as well as 0.7.6)

The generated error is as follows:
```
error[E0425]: cannot find function `__basepri_r` in module `crate::asm::inline`
 --> /Users/janoscserna/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/register/basepri.rs:6:15
  |
6 |     call_asm!(__basepri_r() -> u8)
  |               ^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__basepri_w` in module `crate::asm::inline`
  --> /Users/janoscserna/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/register/basepri.rs:22:19
   |
22 |         call_asm!(__basepri_w(basepri: u8));
   |                   ^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__basepri_max` in module `crate::asm::inline`
  --> /Users/janoscserna/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/register/basepri_max.rs:19:19
   |
19 |         call_asm!(__basepri_max(basepri: u8));
   |                   ^^^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__faultmask_r` in module `crate::asm::inline`
  --> /Users/janoscserna/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/register/faultmask.rs:29:28
   |
29 |     let r: u32 = call_asm!(__faultmask_r() -> u32);
   |                            ^^^^^^^^^^^^^ not found in `crate::asm::inline`

error: invalid register `r0`: unknown register
   --> /Users/janoscserna/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/../asm/inline.rs:197:24
    |
197 |     asm!("bkpt #0xab", inout("r0") nr, in("r1") arg, options(nomem, nostack, preserves_flags));
    |                        ^^^^^^^^^^^^^^

error: invalid register `r1`: unknown register
   --> /Users/janoscserna/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/../asm/inline.rs:197:40
    |
197 |     asm!("bkpt #0xab", inout("r0") nr, in("r1") arg, options(nomem, nostack, preserves_flags));
    |                                        ^^^^^^^^^^^^

For more information about this error, try `rustc --explain E0425`.
error: could not compile `cortex-m` (lib) due to 6 previous errors
warning: build failed, waiting for other jobs to finish...
```

is this a simple dependency I'm missing? or is this an issue?  I couldn't find any issues that seemed related.

adamgreig, 2023-07-04:
Are you definitely building for a thumb target? I think this error often occurs if you try and build for a target like x86; you should use `--target thumbv7em-none-eabi` or similar, or set it in your `.cargo/config.toml` to avoid needing to write it every time.

Takkaryx, 2023-07-04:
ah, you are correct, my rust-toolchain.toml had a type in the target line 🤦

closed 2023-07-04 by Takkaryx