rust-embedded/cortex-m #485: Potential unsound casting from `volatile_register::RW` to the wrapped type

shinmao, 2023-09-26:
Hi, I am wondering whether the casting here is unsound:
https://github.com/rust-embedded/cortex-m/blob/1746a63ca16b68514ea23dcca1543aed00165452/src/peripheral/scb.rs#L172-L174
Based on the definition, the wrapper struct `volatile_register::RW` (let me abbreviate it as `vr` here) didn't have stable layout, and `rustc` would preserve the right to insert padding or reorder the structure. `vr<u32>` should be declared as `transparent`; otherwise, casting it to `u32` could lead to UB such as uninitialized memory exposure to `ptr::read_volatile`.

adamgreig, 2023-09-26:
Yep, you're right. In practice this is very unlikely to lead to any issues but in principle volatile_register should add a `repr` to the structs. There's an open PR to do so here: https://github.com/japaric/volatile-register/pull/9 but it's not merged yet.

Dirbaio, 2023-10-20:
volatile-register v0.2.2 is now out, it adds `#[repr(transparent)]` so the pointer cast is now sound.

commit 8050902f (2023-11-04): Merge pull request #493 from rust-embedded/volatile-reg-0.2.2

cortex-m: update to volatile-register 0.2.2, fixes #485
commit a53f6292 (2023-11-04): cortex-m: update to volatile-register 0.2.2, fixes #485
commit 067522dd (2025-08-31): cortex-m: update to volatile-register 0.2.2, fixes #485 (#493)
closed 2023-11-04 by adamgreig