rust-embedded/cortex-m #498: "Extra Sections" example is potentially unsound

jamesmunns, 2023-11-26:
In the `cortex-m-rt` docs, the [extra sections example](https://docs.rs/cortex-m-rt/latest/cortex_m_rt/#extra-sections) shows the definition of a zero-initialized static mut array in a section outside of RAM.

I believe this to be unsound, and difficult, if not impossible, to use totally soundly.

The CRT is required to initialize all statics, either to their default value (in `.data`), or to zero (in `.bss`). cortex-m-rt explicitly only performs initialization for data and bss sections in the `RAM` region, meaning the `CCRAM` *must* be considered to be uninitialized in the example as currently written.

At the very least if we include an example like this, we should discuss this deficiency, and state that `pre_init` must be used to guarantee that the static is initialized at the start of the program.

I *personally* am of the opionion that we *cannot* soundly perform this initialization in a `pre_init` in Rust, as that means that *Rust code is running without all statics initialized*, which violates the agreement the language makes with the platform environment.

This argument is part of why we switched to an assembly CRT, rather than the previously Rust-based init sequence.

jamesmunns, 2023-11-26:
@Dirbaio noted (and I agree) that changing it to:

```rust
#[link_section=".ccmram.BUFFERS"]
static mut BUF: MaybeUninit<[u8; 1024]> = MaybeUninit::uninit();
```

alleviates the majority of the UB (though, still recommends `static mut` which is still hard to use right IMO, but is not "plainly always UB" anymore)

thejpster, 2023-11-26:
Can you soundly perform the initialisation in `pre_init` in assembly?

jamesmunns, 2023-11-26:
> Can you soundly perform the initialisation in `pre_init` in assembly?

In my opinion, yes.

`__pre_init` is the actual ABI symbol for the function called AFTER other init and BEFORE control transfer to `Rust` code. See [here](https://github.com/rust-embedded/cortex-m/blob/fc2c6cc5c8c3471d1ab22c11051b5664138d1bd8/cortex-m-rt/src/lib.rs#L512-L515).

Assembly has no "abstract machine" requirements - so there isn't any soundness requirement here.

I'm not certain we could use inline assembly within a rust `pre_init` function, e.g. using the [`#[pre_init]`](https://docs.rs/cortex-m-rt/latest/cortex_m_rt/attr.pre_init.html) macro, I believe it *must* be a global asm definition (or externally assembled file that defines the underlying `__pre_init` symbol).

adamgreig, 2024-04-16:
I've updated the documentation in #525 to specify that you have to use `MaybeUninit` or `pre_init`, though I'd be open to adding a more useful/complete example too.