rust-embedded/cortex-m #509: More-helpful linker errors

tomhampshire, 2024-01-30:
Having just debugged an issue, it is clear that it would have helped me if the following asserts gave some clear information as to _why_ "The .text section must be placed inside the FLASH memory.". For example, add to this: "It is likely that this is failing this because you are using more FLASH than you actually have"
https://github.com/rust-embedded/cortex-m/blob/6b3a5b7fb95fe98fb05ddb4270d2d14709afe99c/cortex-m-rt/link.x.in#L255-L262

jamesmunns, 2024-01-30:
Errors as it looks today:

```
  = note: rust-lld: error: 
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'
```

adamgreig, 2024-01-30:
Thanks for reporting this.

I think we should change this assert to just check that `_stext` is inside `FLASH` without checking the whole of `.text` fits, and leave that check to the linker instead (which will give a better error message).

Something like:

```
 ASSERT(_stext > ORIGIN(FLASH) && _stext < ORIGIN(FLASH) + LENGTH(FLASH), " 
 ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory. 
 Set _stext to an address inside the FLASH region.'"); 
```

PR #510 (2024-02-10): cmrt: change condition on _stext location check
This avoids spurious errors when `.text` doesn't fit in `FLASH` which have nothing to do with the position of `_stext`. By not checking, we let the linker provide its own, more-useful, error.

Closes #509.
commit 22ba919b (2024-02-10): cmrt: change condition on _stext location check

This avoids spurious errors when `.text` doesn't fit in `FLASH` which
have nothing to do with the position of `_stext`. By not checking, we
let the linker provide its own, more-useful, error.

Closes #509.
closed 2024-02-10 by adamgreig