rust-embedded/cortex-m #538: `cortex_m::singleton` attribute forwarding is unsound in some cases

jamesmunns, 2024-06-19:
If you use `link_section` to put a singleton in an uninit section, the "has been taken" boolean is also in uninit memory at startup. We read this value which is uninitialized.

https://github.com/rust-embedded/cortex-m/blob/f3f85e683f98d0f44ddadcb725e6e05c51c978b1/cortex-m/src/macros.rs#L72-L74

We could potentially put the `bool` in a separate static that is in the "normal" `.bss` section, but we'd have to document this.

CC @jordens and https://github.com/rust-embedded/cortex-m/pull/522

jordens, 2024-06-19:
And it's especially bad for a bool being the prime example of not just uninit but even init with the wrong value causing UB.

jamesmunns, 2024-06-19:
(fwiw, "link_section" is always an easy way to cause unsoundness in safe code, as with basically any direction you give to the linker, but it FEELS like this is an unexpected outcome rather than an intended one, which we should probably fix and document, or at least document if we say "you can totally use this with link_section".)

jannic, 2024-09-16:
Related: It looks like `link_section` is going to become officially unsafe. https://github.com/rust-lang/rust/issues/129566

Of course, as using `link_section` is very useful and sometimes necessary in embedded contexts, we still should provide documentation on how to use it correctly.

usbalbin, 2024-12-09:
What if the `bool` was split out to its own static which ignores the `link_section` things? Then the `bool` is guaranteed to be valid while the data part is free to be uninit in its wonky linker section

jordens, 2024-12-09:
That was also my plan.