rust-embedded/cortex-m #547: `master` is out of date (0.7.4 vs 0.7.7)

BartMassey, 2024-07-24:
It looks like the release version of `cortex-m` on `crates.io` is 0.7.7, but `master` and `staging` here are 0.7.4. `v0.7.x` seems to contain the current version.

Permission to merge `v0.7.x` onto `master`?

adamgreig, 2024-07-25:
The reason is that `master` is tracking a few breaking changes, so it will eventually be published as 0.8, but we don't bump the version until publishing. In the meantime, several backwards-compatible fixes (and some backports) were made in the `0.7.x` branch and released as new 0.7 versions, but nothing that's not also present in master.

This is confusing but is currently how we generally handle working towards a new breaking-change version. We do have a plan to revamp cortex-m a bit before 0.8 to reduce the impact of breaking changes - the reason we haven't already released 0.8 is that currently it breaks the whole cortex-m ecosystem when we do (new PACs, new HALs, new drivers, etc) and requires/usually gets a semver-hack release in the old version, but that gets harder and harder to do with each release, hence the plan to make it less ecosystem-breaking.

BartMassey, 2024-07-25:
Thanks much for the explanation!

closed 2024-07-25 by BartMassey