rust-embedded/cortex-m #583: Hard fault on debug compilation

kevinbube, 2025-03-23:
Hello all,
I am experimenting with an STM32F303 and try to enter unprivileged mode with program stack pointer set. I use the following minimal example:

```
#![no_std]
#![no_main]

use panic_halt as _;
use cortex_m_rt::entry;
use cortex_m_semihosting::hprintln;
use stm32f3xx_hal::prelude::*;
                    
#[entry]
fn entry() -> ! {
    extern "C" {
        static _user_stack_start: u32;
    }

    // set unprivileged stack
    unsafe {
        let stack_addr = &_user_stack_start as *const u32 as u32;
        cortex_m::register::psp::write(stack_addr)
    };
    
    let mut ctrl_reg = cortex_m::register::control::read();
    ctrl_reg.set_npriv(cortex_m::register::control::Npriv::Unprivileged);
    ctrl_reg.set_spsel(cortex_m::register::control::Spsel::Psp);

    // enter unprivileged mode
    unsafe { cortex_m::register::control::write(ctrl_reg) };

    main();
}

fn main() -> ! {
    hprintln!("hello");

    loop {}
}
```
On a debug build this causes a hard fault at control::write(). On release build it works fine. I guess this is because the compiler does not inline the call on debug and the function crashes on return as it fails to pop the return address from the stack, which changed from MSP to PSP.

Maybe #[inline(always)] is necessary for control::write()?

jannic, 2025-04-15:
Even if `#[inline(always)]` helps in this case (I didn't try it), it can't be a correct solution, as you can't rely on inlining for correctness: https://doc.rust-lang.org/stable/reference/attributes/codegen.html?highlight=inline#the-inline-attribute clearly documents that the compiler is not required to actually inline the function, even with `#[inline(always)]`.

The conclusion may be that you can't change the stack using `cortex_m::register::control::write`.

BartMassey, 2025-04-15:
Needs a macro, sounds like.

adamgreig, 2025-04-15:
Hm, yea, maybe a macro-by-example could emit the inline asm needed. That's probably nicer than just saying everyone has to use asm directly. It would still have to be used pretty carefully though.

Certainly I don't see how we can usefully have this method as a normal function today, and now that inline asm is stable the justification from the time ("it's impossible to do this otherwise in stable rust; at least the function works in release mode") doesn't hold up.

jonathanpallant, 2025-04-16:
I need this to run a particular proprietary RTOS. I suspect the correct solution is an argument given to the `entry` attribute macro.

PR #594 (2025-08-31): Add enter_unprivileged() function
This adds a function to switch to program stack and unprivileged mode.

Fixes #583
commit 539bce12 (2025-08-31): Add enter_unprivileged() function

This adds a function to switch to program stack and unprivileged mode.

Fixes #583
closed 2025-08-31 by adamgreig