rust-embedded/cortex-m #592: Lots of clippy lints

jonathanpallant, 2025-04-16:
Lots of lints seen in https://github.com/rust-embedded/cortex-m/pull/590/checks?check_run_id=40591927014. The codebase could use a tidy up.

jannic, 2026-09-15:
This seems to be fixed. (Likely many by you, judging from the commit messages containing the word "clippy").

Unfortunately, the job results from your link are no longer available. But I can't reproduce any clippy lints currently.
If there are some left that only show up with certain settings (e.g. on a specific target), please re-open the ticket with more details.

closed 2026-09-15 by jannic

jonathanpallant, 2026-09-17:
Yeah https://github.com/rust-embedded/cortex-m/actions/runs/35010057467/job/104519479720 seems OK now