rust-embedded/cortex-m #599: Run program from SRAM

phoracek, 2025-06-08:
Hello,

Is it at all possible to run programs built with cortex-m-rt from anything other than FLASH? More specifically from SRAM of the STM32H750IB? Asking to see if it is worth going down the linker script rabbit hole, or if it is simply not supported by this crate. Thanks!

...

I'm using the Daisy Seed development platform. Their boards use the small 128 kB STM32H7 chips. However, they compensate for it by providing a [bootloader](https://electro-smith.github.io/libDaisy/md_doc_2md_2__a7___getting-_started-_daisy-_bootloader.html). You would first copy the firmware to on an-board QSPI flash storage. The bootloader then copies the firmware to SRAM on boot and runs the program from there. Effectively increasing the maximum firmware size from 128 kB to 512 kB.

The linker script their C++ based SDK uses is available here https://github.com/electro-smith/libDaisy/blob/faddc7402a0e91729cf3dc49e43623a3553159d8/core/STM32H750IB_sram.lds. It places `.text` and `.isr_vector` to SRAM.

I tried using it with my Rust program based on cortex-m and stm32h7xx-hal. I added an alias `REGION_ALIAS(RAM, DTCMRAM);
` to overcome the first compilation issue, but sadly it even then fails:

```
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'

          rust-lld: error:
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'

          rust-lld: error:
          ERROR(cortex-m-rt): The .text section must be placed inside the FLASH memory.
          Set _stext to an address smaller than 'ORIGIN(FLASH) + LENGTH(FLASH)'

          rust-lld: error: section .text virtual address range overlaps with .text
          >>> .text range is [0x24000000, 0x240190AF]
          >>> .text range is [0x24000000, 0x24000073]

          rust-lld: error: section .text load address range overlaps with .text
          >>> .text range is [0x24000000, 0x240190AF]
          >>> .text range is [0x24000000, 0x24000073]
``` 

Best regards,
Petr

phoracek, 2025-06-09:
That's brilliant, thanks a lot @eulerdisk for taking the time to write that down. I'll try it out and report back

eulerdisk, 2025-06-10:
@phoracek Actually, if you don't need further customization and the standard `link.x` provided by `cortex-m-rt` works for you, then you can probably get by with just modifying the `memory.x` like this:

```
MEMORY
{
  RAM : ORIGIN = 0x20000000, LENGTH = 64K
}

REGION_ALIAS("FLASH", RAM);
```

phoracek, 2025-06-10:
@eulerdisk that' much neater, thanks so much! In case any Daisy users stumble upon this - I summarized the process together with an example specific to the Daisy platform here: https://github.com/zlosynth/daisy/tree/main/examples/bootloader